gusmanb/logicanalyzer #170: Please include CLCapture in the LogicAnalyser package

UffeJakobsen, 2024-12-19:
Please include CLCapture in the LogicAnalyser package

It feels a little stupid to have 2 almost identical installations - just to have both GUI and CLI
Also it is a little cumbersome having to disect the 2 packages in order to merge them to one unified package

gusmanb, 2024-12-19:
Usually the users that need CLCapture do not use the GUI app and vice-versa, what is your use case that needs both apps?

UffeJakobsen, 2024-12-19:
For some projects this UI works - for other projects I need to feed the capture into PulseView in a automated manner and do my analysis there. 

Imagine repetitive UI operations such at this: capture in one UI - export - (optional) preprocessing - import into PulseView - do some actual work.... after repeating this many times you wish for something more automated.

UffeJakobsen, 2024-12-19:
The last dissection of 5.1 packages came up with 5 files that needed to be copies from CLcapture and into the LogicAnalyser installation:

 - CLCapture.runtimeconfig.json
 - CLCapture.pdb
 - CLCapture.dll
 - CLCapture.deps.json
 - CLCapture

Their total size is around 400KB - so I guess that putting them into the LogicAnalyser package wouldn't be noticed by anyone not needing it - as the package already is huge (110MB unpacked)

UffeJakobsen, 2024-12-19:
> Have you tried V6.0? It includes all the analyzers from Sigrok. You can download the betas from the discussion sections.
> 

No, I'm a little out of time for the moment...

PS: just downloaded 6.0.0-beta5 (I believe that is the latest) and gave it a quick run on the hardware available - and it instantly core dumped before presenting any UI - failing on to access `/sys/bus/usb/devices/1-9/serial`:

PPS: there is no pico attached to this system (I'm at $WORK) - but on 5.1 it was possible to work in the UI with SDL etc - without any actual hardware available.

Complete output:
```
Unhandled exception. System.IO.FileNotFoundException: Could not find file '/sys/bus/usb/devices/1-9/serial'.
File name: '/sys/bus/usb/devices/1-9/serial'
   at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirError)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String path, OpenFlags flags, Int32 mode, Boolean failForSymlink, Boolean& wasSymlink, Func`4 createOpenException)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, UnixFileMode openPermissions, Int64& fileLength, UnixFileMode& filePermissions, Boolean failForSymlink, Boolean& wasSymlink, Func`4 createOpenException)
   at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, Nullable`1 unixCreateMode)
   at System.IO.StreamReader.ValidateArgsAndOpenPath(String path, Encoding encoding, Int32 bufferSize)
   at System.IO.File.ReadAllText(String path, Encoding encoding)
   at SharedDriver.DeviceDetector.DetectLinux() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\SharedDriver\DeviceDetector.cs:line 70
   at SharedDriver.DeviceDetector.Detect() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\SharedDriver\DeviceDetector.cs:line 40
   at LogicAnalyzer.MainWindow.RefreshPorts() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\MainWindow.axaml.cs:line 1292
   at LogicAnalyzer.MainWindow..ctor() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\MainWindow.axaml.cs:line 147
   at LogicAnalyzer.App.OnFrameworkInitializationCompleted() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\App.axaml.cs:line 24
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.SendOrPostCallbackDispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.X11.X11PlatformThreading.CheckSignaled()
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, Action`1 lifetimeBuilder)
   at LogicAnalyzer.Program.Main(String[] args) in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\Program.cs:line 22
zsh: IOT instruction (core dumped)  ./LogicAnalyzer
./LogicAnalyzer  1,20s user 0,15s system 95% cpu 1,419 total

```

gusmanb, 2024-12-19:
Thanks for the report, that's the new feature for autodetection and is not well tested under Linux. I will update the packages today.

gusmanb, 2025-02-01:
New terminal capture app has been added and an "all-in-one" package has been added to the release.

closed 2025-02-01 by gusmanb