gusmanb/logicanalyzer #171: RPI5 RP1 PIOlib support?

Ragebone, 2024-12-21:
Hello,
with the recent release of the PIOlib, the PIO blocks inside the RP1 of the Raspberry Pi 5 are starting become usable.
I am wondering what your ideas and plans are for this in the future.

Link for reference: https://www.raspberrypi.com/news/piolib-a-userspace-library-for-pio-control/

As they state, the roughly 1us latency on the round-trip makes me question the usability of this with things like your burst triggers and anything where you have to react to an event.
On the other hand, the bandwidth of the RP1 could  be enough for streaming at decent sampling rates.

What are your thoughts? 

Cheers
Norman

gusmanb, 2024-12-21:
As far as I know, they cannot be used like I do on the pico, so for now no plans.

First of all, I cannot overclock the unit to the 400Mhz that I'm reaching right now, so the speed would be limited.

Next the complex triggers cannot be implemented as the RP1 has only a single state machine and I need two because the programs fill the 32 instruction memory of the two units on the pico, so only edge triggers and no chaining would be possible limiting a lot the functionality.

Finally the DMAs I'm not sure the real speed that can be achieved and the library only allows to do direct transfers, no ring buffers like I do in the pico to have pre and post samples.

closed 2024-12-21 by gusmanb

Ragebone, 2024-12-21:
Bummer, i didn't expect the one PIO block to be that limiting. 
Appreciate your response and thoughts! 
Keep up the good work.
Thank you very much!

Cheers