gusmanb/logicanalyzer #186: "Drill Map" marking on PCB

czabel-personal, 2025-02-01:
Hi @gusmanb - I just went to order a few 6.0 analyzers on PCBWay, but noticed an odd section at the bottom of the PCB that has the marking "Drill map:" on both sides. Is this intended, or do you want to update your design on their website to fix this? Or, does it matter?

gusmanb, 2025-02-01:
I think that is something that PCBWay added, I have used these same files with JLCPCB multiple times with no problems.

czabel-personal, 2025-02-01:
Great, then I'll place my order and not worry about it. Thanks for all your work on this, I'm excited to try it out!

closed 2025-02-01 by czabel-personal

gusmanb, 2025-02-01:
If you find any issue please notify me, this is the first time I've used PCBWay. Thanks in advance.

gusmanb, 2025-02-01:
Hey, wait, I think I have found the problem and is a discrepancy on what PCBWay supports, in JLCPCB they don't need the edge cuts. Let me upload a new version.

gusmanb, 2025-02-01:
New version uploaded. Now it shows correctly.