gusmanb/logicanalyzer #187: Jason Fork with 220ohm ?

hgomez, 2025-02-02:
I received Jason Fork PCB a few days ago and assembled one. V6.0 was released at that time so I used 6.0 firmware with pico and pico2.

Sadly, when plugged to the board, pico and pico2 didn't worked, no even seen by PC. 

Chips were hot so I didn't go further.

There was no problem with pico and pico2 when unplugged hopefully.

I reviewed Jason schematic and was wondering about the 220ohms resistors. 15mA by GPIO, 360mA in total.

I found this article who warns about 220ohms, too small

https://grantwinney.com/raspberry-pi-using-pullup-and-pulldown-resistors/

I'll change to 10k resistors and have a new try.

gusmanb, 2025-02-02:
Hi.

First of all, Jason's design will not work at all with the Pico 2, it does not have push-pull transceivers nor the proper pulldown for the trigger pin, so the IOs will get stuck.

Second, the 220Ohm res are a protection in case of failure and a limitation if you use signals with more than 3.3v, it does not mean that it will consume 15mA per IO, when the GPIOs are configured as input they are high impedance, only the leakage current will flow at them and usually will be on the nA range.

If the device is getting really hot then it looks more like you have a short somewhere, maybe between +3v3 and +5v.

If you want to use a Pico 2 you need the new design, that's why I created it.

hgomez, 2025-02-02:
I'll use your design, I already ordered PCB 😉
I'll have a closer look to shorts

closed 2025-02-02 by hgomez