gusmanb/logicanalyzer #188: Create DMG for Mac OS

codefhacker, 2025-02-05:
Hi,

First, many thanks for your excellent work on this piece of software and hardware.

I was wondering if it's possible to create a DMG for the logic analyzer program?

I tried doing this myself, but I keep getting the following error:

`Cannot load Sigrok decoders. Make sure Python is installed on your computer. If, despite being installed, you still have problems, you can specify the path to the Python library in "python.cfg".`

What I did was create a LogicAnalyzer.app folder and placed all the files inside the Contents folder. The app starts, but it gives the error mentioned above.

Currently, on macOS, when I want to run the logic analyzer program, I have to locate the executable and run it manually. It would be much easier to launch it like a regular app using a DMG or a .app file.

Thanks for your help!

gusmanb, 2025-02-15:
Sorry but my knowledge of MacOS is very limited, I have no clue how a DMG or app works and from the message I suspect that may have some kind of storage isolation and is not able to find Python.

closed 2025-02-15 by gusmanb

codefhacker, 2025-02-15:
No problem thank you for searching it out anyways.

If I find it out myself I will post it here so others can reproduce it.