gusmanb/logicanalyzer #199: error loading decoders

LE1ED4523, 2025-03-02:
> Cannot load Sigrok decoders. Make sure Python is installed on your computer. If, despite being installed, you still have problems, you can specify the path to the Python library in "python.cfg".

there is no "python.cfg" I only find this log.

[PythonInitLog.txt](https://github.com/user-attachments/files/19041525/PythonInitLog.txt)

gusmanb, 2025-03-02:
Just create one and put the path to the library. Also, you only have python
as part of msys, you need a regular standalone installation.

rsipka, 2025-03-04:
I was trying on Linux, but I'm at my wits end.
By default, on Fedora 41 the python is 3.13, so I had to install 3.12, and I need to configure the python.cfg file. However, when trying to configure the library path, what is the DLL on Linux? I saw /usr/bin/python3.12 is linking  /lib64/libpython3.12.so.1.0 but that doesn't work (tried also python3.8 library). 

Based on documentation (https://github.com/pythonnet/pythonnet) it seems that should indeed be the correct library to link.

gusmanb, 2025-03-05:
> I was trying on Linux, but I'm at my wits end. By default, on Fedora 41 the python is 3.13, so I had to install 3.12, and I need to configure the python.cfg file. However, when trying to configure [...]

Usually that's not the correct library, for example for Python 3.7 it looks like this: “/usr/lib/python3.7/config-3.7m-x86_64-linux-gnu/libpython3.7m.so“

Try this: remove the python.cfg file, let the app run, and if it still fails send the "PythonInit.log" file that will be created, it will contain debug info and from it I may be able to determine why it does not work.

rsipka, 2025-03-05:
On Fedora, the `/lib` and `/lib64` are linked to `/usr/lib` and `/usr/lib64`. 
There is nothing under `/usr/lib64/python3.12/config-3.12-x86_64-linux-gnu`

The `/usr/lib64/libpython3.12.so.1.0` is contained in `python3.12-libs` rpm package, and there isn't other shared library.

### Here is test when having configured `python.cfg`:
```3/5/2025 8:41 AM -> Initializing python system...
3/5/2025 8:41 AM -> Reading python path from config file...
3/5/2025 8:41 AM -> Stablished path: /usr/lib64/libpython3.12.so.1.0

3/5/2025 8:41 AM -> Initializing decoders...
3/5/2025 8:41 AM -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99
```

### Here is test when having **NO** `python.cfg`:
```3/5/2025 8:45 AM -> Initializing python system...
3/5/2025 8:45 AM -> Initializing python installation detection...
3/5/2025 8:45 AM -> Target Interpreter: python,  Initializing Python Installation Detection...
3/5/2025 8:45 AM -> Target Interpreter: python,  Python version found: 313
3/5/2025 8:45 AM -> Target Interpreter: python,  Python version parsed: 3.13
3/5/2025 8:45 AM -> Target Interpreter: python,  PythonExecutable, Finding python library on Mac/Linux...
3/5/2025 8:45 AM -> Target Interpreter: python,  PythonExecutable, Script output: 
3/5/2025 8:45 AM -> Target Interpreter: python,  Final python path: 
3/5/2025 8:45 AM -> Target Interpreter: python,  Cannot retrieve Python library path, aborting.
3/5/2025 8:45 AM -> Target Interpreter: python3,  Initializing Python Installation Detection...
3/5/2025 8:45 AM -> Target Interpreter: python3,  Python version found: 313
3/5/2025 8:45 AM -> Target Interpreter: python3,  Python version parsed: 3.13
3/5/2025 8:45 AM -> Target Interpreter: python3,  PythonExecutable, Finding python library on Mac/Linux...
3/5/2025 8:45 AM -> Target Interpreter: python3,  PythonExecutable, Script output: 
3/5/2025 8:45 AM -> Target Interpreter: python3,  Final python path: 
3/5/2025 8:45 AM -> Target Interpreter: python3,  Cannot retrieve Python library path, aborting.
3/5/2025 8:45 AM -> Valid installations found: 0
3/5/2025 8:45 AM -> No valid python installation found, aborting startup.
```

I will try to dig a bit more into this

gusmanb, 2025-03-05:
Ok, I have downloaded an image of Fedora 41 and python on it is a bit of a mess... First, all the variables point to inexistent files, I use python itself to determine the library to load using the 'LDLIBRARY' config setting, it reports that the library is "libpython.3.13.so" and this does not exist in the system, so the autodetection will never work...

So yes, your file path seems to be correct (and yes, it should work with Python 3.13 also, I didn't notified about it as I didn't had enough time to validate it in Linux/Mac, only tested it under Windows).

Next, digging a bit I have found more users with other programs that use pythonnet and Fedora having what possibly is the same problem, it seems that there is a missing dependency (sigh...).

I have tried the workaround but still the library refuses to load, it seems there are more missing dependencies but I have no clue on which ones are missing, the only thing that I get in the logs is `Could not load libpython3.13.so.1.0
 with flags RTLD_NOW | RTLD_GLOBAL: libpython3.13.so.1.0
: cannot open shared object file: No such file or directory` which seems to be caused by missing dependencies...

The only thing that I can think that maybe could work is to create a docker container with another distro and try with it...

rsipka, 2025-03-06:
**SOLVED**: it was missing `python3-devel` rpm package.

I noticed that on my laptop (where I have also Fedora 41) it works.

There was one more file in `/usr/lib64/` on the laptop:
```
/usr/lib64/libpython* 
lrwxrwxrwx. 1 root root      20  4. úno 01.00 /usr/lib64/libpython3.13.so -> libpython3.13.so.1.0
-rwxr-xr-x. 1 root root 5683480  4. úno 01.00 /usr/lib64/libpython3.13.so.1.0
-rwxr-xr-x. 1 root root   15312  4. úno 01.00 /usr/lib64/libpython3.so
```

vs PC:
```
/usr/lib64/libpython* 
-rwxr-xr-x. 1 root root 5683480  4. úno 01.00 /usr/lib64/libpython3.13.so.1.0
-rwxr-xr-x. 1 root root   15312  4. úno 01.00 /usr/lib64/libpython3.so
```

The symbolic link `/usr/lib64/libpython3.13.so -> libpython3.13.so.1.0` was missing from the PC where it couldn't find Python.

I've checked which package provides the file, and:
```
rpm -qf /usr/lib64/libpython3.13.so
python3-devel-3.13.2-1.fc41.x86_64
```

After installing the package on the original PC, LogicAnalyzer **will find & use Python 3.13 without a problem**. 

I've tried to use `python.cfg` (instead of auto-find) when having the `python3-devel` installed, and it fails to load/find python when specifying either of:
- `/usr/lib64/libpython3.13.so.1.0`
- `/usr/lib64/libpython3.13.so`

Maybe the manual config doesn't work well under linux?

NWSpitfire, 2025-03-06:
Mine is not completely relevant as I am using Windows 11. But I was having this issue, and it turns out it was autodetecting a Python instance that installed with Inkscape - instead of my installed Python 3.10.

I specified my path to the Python310.dll in my C: drive and now it works perfectly.

Thanks

```
06/03/2025 17:03 -> Initializing python system...
06/03/2025 17:03 -> Initializing python installation detection...
06/03/2025 17:03 -> Target Interpreter: python,  Initializing Python Installation Detection...
06/03/2025 17:03 -> Target Interpreter: python,  Python version found: 311
06/03/2025 17:03 -> Target Interpreter: python,  Python version parsed: 3.11
06/03/2025 17:03 -> Target Interpreter: python,  Finding python library on Windows...
06/03/2025 17:03 -> Target Interpreter: python,  Script output: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:03 -> Target Interpreter: python,  Final python path: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:03 -> Target Interpreter: python,  Python path found: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:03 -> Target Interpreter: python3,  Initializing Python Installation Detection...
06/03/2025 17:04 -> Target Interpreter: python3,  Python version found: 310
06/03/2025 17:04 -> Target Interpreter: python3,  Python version parsed: 3.10
06/03/2025 17:04 -> Target Interpreter: python3,  Finding python library on Windows...
06/03/2025 17:04 -> Target Interpreter: python3,  Script output: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
06/03/2025 17:04 -> Target Interpreter: python3,  Final python path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
06/03/2025 17:04 -> Target Interpreter: python3,  Python path found: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
06/03/2025 17:04 -> Valid installations found: 2
06/03/2025 17:04 -> Selected version: 3.11
06/03/2025 17:04 -> Stablished path: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:04 -> Initializing decoders...
06/03/2025 17:04 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99
06/03/2025 17:27 -> Initializing python system...
06/03/2025 17:27 -> Initializing python installation detection...
06/03/2025 17:27 -> Target Interpreter: python,  Initializing Python Installation Detection...
06/03/2025 17:27 -> Target Interpreter: python,  Python version found: 311
06/03/2025 17:27 -> Target Interpreter: python,  Python version parsed: 3.11
06/03/2025 17:27 -> Target Interpreter: python,  Finding python library on Windows...
06/03/2025 17:27 -> Target Interpreter: python,  Script output: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:27 -> Target Interpreter: python,  Final python path: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:27 -> Target Interpreter: python,  Python path found: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:27 -> Target Interpreter: python3,  Initializing Python Installation Detection...
06/03/2025 17:27 -> Target Interpreter: python3,  Python version found: 310
06/03/2025 17:27 -> Target Interpreter: python3,  Python version parsed: 3.10
06/03/2025 17:27 -> Target Interpreter: python3,  Finding python library on Windows...
06/03/2025 17:27 -> Target Interpreter: python3,  Script output: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
06/03/2025 17:27 -> Target Interpreter: python3,  Final python path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
06/03/2025 17:27 -> Target Interpreter: python3,  Python path found: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
06/03/2025 17:27 -> Valid installations found: 2
06/03/2025 17:27 -> Selected version: 3.11
06/03/2025 17:27 -> Stablished path: C:\Program Files\Inkscape\Python311.dll
06/03/2025 17:27 -> Initializing decoders...
06/03/2025 17:27 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
[... cut ...]
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99
06/03/2025 17:48 -> Initializing python system...
06/03/2025 17:48 -> Reading python path from config file...
06/03/2025 17:48 -> Stablished path: C:\Program Files\Python310\python310.dll
06/03/2025 17:48 -> Initializing decoders...
06/03/2025 17:48 -> Python initialization completed.
```

gusmanb, 2025-03-06:
> Mine is not completely relevant as I am using Windows 11. But I was having this issue, and it turns out it was autodetecting a Python instance that installed with Inkscape - instead of my installed Python 3.10.
> 
> I created the python.cfg file inside the LA Software folder, specified my path to the Python310.dll on my C: drive and now it works perfectly.
> 
> Thanks
> 
> ```
> 06/03/2025 17:03 -> Initializing python system...
> 06/03/2025 17:03 -> Initializing python installation detection...
> 06/03/2025 17:03 -> Target Interpreter: python,  Initializing Python Installation Detection...
> 06/03/2025 17:03 -> Target Interpreter: python,  Python version found: 311
> 06/03/2025 17:03 -> Target Interpreter: python,  Python version parsed: 3.11
> 06/03/2025 17:03 -> Target Interpreter: python,  Finding python library on Windows...
> 06/03/2025 17:03 -> Target Interpreter: python,  Script output: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:03 -> Target Interpreter: python,  Final python path: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:03 -> Target Interpreter: python,  Python path found: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:03 -> Target Interpreter: python3,  Initializing Python Installation Detection...
> 06/03/2025 17:04 -> Target Interpreter: python3,  Python version found: 310
> 06/03/2025 17:04 -> Target Interpreter: python3,  Python version parsed: 3.10
> 06/03/2025 17:04 -> Target Interpreter: python3,  Finding python library on Windows...
> 06/03/2025 17:04 -> Target Interpreter: python3,  Script output: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
> 06/03/2025 17:04 -> Target Interpreter: python3,  Final python path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
> 06/03/2025 17:04 -> Target Interpreter: python3,  Python path found: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
> 06/03/2025 17:04 -> Valid installations found: 2
> 06/03/2025 17:04 -> Selected version: 3.11
> 06/03/2025 17:04 -> Stablished path: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:04 -> Initializing decoders...
> 06/03/2025 17:04 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
>    at Python.Runtime.Runtime.Py_IsInitialized()
>    at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
>    at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
>    at Python.Runtime.Runtime.Initialize(Boolean initSigs)
>    at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
>    at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
>    at Python.Runtime.PythonEngine.Initialize()
>    at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99
> 06/03/2025 17:43 -> Initializing python system...
> 06/03/2025 17:43 -> Initializing python installation detection...
> 06/03/2025 17:43 -> Target Interpreter: python,  Initializing Python Installation Detection...
> 06/03/2025 17:43 -> Target Interpreter: python,  Python version found: 311
> 06/03/2025 17:43 -> Target Interpreter: python,  Python version parsed: 3.11
> 06/03/2025 17:43 -> Target Interpreter: python,  Finding python library on Windows...
> 06/03/2025 17:43 -> Target Interpreter: python,  Script output: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:43 -> Target Interpreter: python,  Final python path: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:43 -> Target Interpreter: python,  Python path found: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:43 -> Target Interpreter: python3,  Initializing Python Installation Detection...
> 06/03/2025 17:43 -> Target Interpreter: python3,  Python version found: 310
> 06/03/2025 17:43 -> Target Interpreter: python3,  Python version parsed: 3.10
> 06/03/2025 17:43 -> Target Interpreter: python3,  Finding python library on Windows...
> 06/03/2025 17:43 -> Target Interpreter: python3,  Script output: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
> 06/03/2025 17:43 -> Target Interpreter: python3,  Final python path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
> 06/03/2025 17:43 -> Target Interpreter: python3,  Python path found: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\Python310.dll
> 06/03/2025 17:43 -> Valid installations found: 2
> 06/03/2025 17:43 -> Selected version: 3.11
> 06/03/2025 17:43 -> Stablished path: C:\Program Files\Inkscape\Python311.dll
> 06/03/2025 17:43 -> Initializing decoders...
> 06/03/2025 17:43 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
>    at Python.Runtime.Runtime.Py_IsInitialized()
>    at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
>    at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
>    at Python.Runtime.Runtime.Initialize(Boolean initSigs)
>    at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
[... cut ...]
Yes, that's why I added the "python.cfg", python installs are a bit of... convoluted let's say..., you can have many in your machine and to detect which one is the appropiated is nearly impossible, the routine that I implemented what tries to do is to find the newest one and use it, that's why in your case it choosed 3.11 instead of the 3.10 one.

I would love to be able to use python embedded, but as far as I know it is only available for Windows, Linux and MacOS forces you to have a shared install, is a shame because it would remove all these problems.

Dgarc359, 2025-04-06:
I had this issue on a fresh windows install. I was able to resolve it by installing python 3.10 and ensuring that 'add python to path' was checked.

softfoot, 2026-01-28:
A word of warning - the LA documentation says that it needs python313. I had problems with 314 - but 3.13.11 works fine.
My python.cfg contains the following as a single line with no trailing newline and no quotes :-

C:\Users\Bob\AppData\Local\Programs\Python\Python313\python313.dll

Does any one know if you can have multiple versions of python installed at the same time ?? I have an app that needs 314 :-(
Dave

petrm, 2026-02-21:
Actually, letting the application autodetect system python works. It found /Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/libpython3.9.dylib using /usr/bin/python3.