gusmanb/logicanalyzer #206: Allow to create more burst by disabling the measurement feature

gusmanb, 2025-03-06:
Allow to use up to 65534 bursts of captures if burst measure is disabled.

almarush, 2026-03-11:
Dear Dr. Gusman, is it possible to increase the number of samples if I use RP2350 together with W25Q128JV 128Mbit 16MB memory chip?

gusmanb, 2026-03-11:
I'm on the process of defining a new version that will use one QSPI RAM, but this still limits a lot the maximum sampling bandwith, internal RAM is able to move around 1.2GB of data per second meanwhile the maximum I've achieved with PSRAM is 51MB/s, so we are talking about 24 times slower in the best case, in practice this will limit to 10/20Mhz the sampling frecuency.

almarush, 2026-03-11:
Thank you for your reply. I work a lot with old computer equipment, and the signal speed rarely exceeds 10 MHz. But there are a lot of signals that need to be measured. Your work is invaluable. Your logic analyzer is very useful. Thank you.