gusmanb/logicanalyzer #220: logicanalyzer_6.0.0.1-win-x64 open error

mz1874, 2025-03-14:
My local python verison is 3.12， I configured the python.cfg to my lib like blow 
`C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\python312.dll`
or like below 
`D:\Softwares\anocanda\python312.dll`

It still not works , here is the logs you can check 

`2025/3/14 21:03 -> Initializing python system...
2025/3/14 21:03 -> Initializing python installation detection...
2025/3/14 21:03 -> Target Interpreter: python,  Initializing Python Installation Detection...
2025/3/14 21:03 -> Target Interpreter: python,  Python version found: 312
2025/3/14 21:03 -> Target Interpreter: python,  Python version parsed: 3.12
2025/3/14 21:03 -> Target Interpreter: python,  Finding python library on Windows...
2025/3/14 21:03 -> Target Interpreter: python,  Script output: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Python312.dll
2025/3/14 21:03 -> Target Interpreter: python,  Final python path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Python312.dll
2025/3/14 21:03 -> Target Interpreter: python,  Python path found: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Python312.dll
2025/3/14 21:03 -> Target Interpreter: python3,  Initializing Python Installation Detection...
2025/3/14 21:03 -> Target Interpreter: python3,  Python version found: 312
2025/3/14 21:03 -> Target Interpreter: python3,  Python version parsed: 3.12
2025/3/14 21:03 -> Target Interpreter: python3,  Finding python library on Windows...
2025/3/14 21:03 -> Target Interpreter: python3,  Script output: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Python312.dll
2025/3/14 21:03 -> Target Interpreter: python3,  Final python path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Python312.dll
2025/3/14 21:03 -> Target Interpreter: python3,  Python path found: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Python312.dll
2025/3/14 21:03 -> Valid installations found: 2
2025/3/14 21:03 -> Selected version: 3.12
2025/3/14 21:03 -> Stablished path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\Python312.dll
2025/3/14 21:03 -> Initializing decoders...
2025/3/14 21:03 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99
2025/3/14 21:08 -> Initializing python system...
2025/3/14 21:08 -> Reading python path from config file...
2025/3/14 21:08 -> Stablished path: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2544.0_x64__qbz5n2kfra8p0\python312.dll
2025/3/14 21:08 -> Initializing decoders...
2025/3/14 21:08 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99`

gusmanb, 2025-03-14:
You have installed Python using a Windows Store package, try to install it with the regular distribution that you can find in https://www.python.org/

mz1874, 2025-03-14:
> You have installed Python using a Windows Store package, try to install it with the regular distribution that you can find in https://www.python.org/

PLS, do not close the issue yet , let retry again , as i tired to do that in a new computer .  I will reinstall python though the python.org

gusmanb, 2025-03-14:
Sure, I do not close issues until the problem is confirmed to be solved or is clearly not a problem :)

mz1874, 2025-03-14:
> Sure, I do not close issues until the problem is confirmed to be solved or is clearly not a problem :)

Yes , it works . however in the logicAnalyzer windows there are no any channels . no error logs 

[image]

mz1874, 2025-03-14:
> Have you captured anything?

No, sir , I enabled the channels 17 to 24 and connected one of the channel to 3v3. There are nothing in the scrren . I mean the channel lines when capturing

mz1874, 2025-03-14:
like this 

[image]

gusmanb, 2025-03-14:
No, capture data, not "start" a capture, you will see the data once the capture is completed. To see anything the trigger must be met and then the data will be captured by the device and then sent to the application. This is a logic analyzer, not an oscilloscope :)

gusmanb, 2025-03-14:
https://github.com/gusmanb/logicanalyzer/wiki/06---The-LogicAnalyzer-program#running-the-capture

closed 2025-03-14 by mz1874