gusmanb/logicanalyzer #221: Device reported error starting capture. Restart the device and try again.

ealvarezd, 2025-03-16:
I get this error when I try to capture data. The program correctly recognizes the card. When I click "Capture," it displays the window for selecting channels. However, when I click "Accept," I get the error "Device reported error starting capture. Restart the device and try again."

gusmanb, 2025-03-16:
Need more info:

Software version.
Firmware version.
Device version.
Do this always happens or after one capture?

Anything else that could give more light also.

ealvarezd, 2025-03-16:
Greetings... I was about to change the title... I was using firmware V6.5.1 with software V6.0.0.1... and I got the error... I tried with software version 6.5.0 and now the error doesn't appear.

closed 2025-03-16 by ealvarezd

gusmanb, 2025-03-16:
Beware, V6.5 is in development and is not supported.

ealvarezd, 2025-03-16:
But should firmware v6.5 work with software v6.0.0.1? If I use those versions, I get the error... using Windows 10 x64, and tested on two different computers. Software v6.0.0.1 detects the card correctly, the information shows everything correctly, and allows me to configure the capture parameters, but when I click "Accept," I get the error.

gusmanb, 2025-03-16:
No, firmware and software 6.5 and are under development and are not backwards compatible, you should not use none of those, only v6.0/6.0.1 are supported.

ealvarezd, 2025-03-16:
The link in the READ.ME sends me to a page on Mega and the only version that appears is 6.5.

gusmanb, 2025-03-16:
The link in the read.me sends to the releases section, not to mega...

[image]

ealvarezd, 2025-03-16:
[image]

gusmanb, 2025-03-16:
So you went to an old link about when V6.0 wasn't released and downloaded it form there... and to make it worse you had the v6.0.0.1 software version that you downloaded from the releases section... right?

ealvarezd, 2025-03-16:
The version I have is this one: "all-in-one _6.0.0.1-win-x64" and the firmware, which I downloaded from the Mega link is: "logic-analyzer-firmware_v6.5.1_BOARD_PICO"
and that combination gives me an error... the question is: where is the 6.0.0.1 firmware?

gusmanb, 2025-03-16:
6.0.0.1 is a minor review of the software only, the firmware is 6.0.0.0 and is in the same place, in the releases section: https://github.com/gusmanb/logicanalyzer/releases/tag/v6.0.0.0

ealvarezd, 2025-03-16:
Is this your GitHub page?

[image]

It's the one I've been visiting, and none of the read.me links take me to the pages you've shown me here. Something's wrong with those links.

ealvarezd, 2025-03-17:
[image]

This is beautiful. I really appreciate your time. You've done a fantastic job creating this analyzer.
Thank you so much.

gusmanb, 2025-03-17:
Is the same page were you downloaded 6.0.0.1, scroll down and it's there.