gusmanb/logicanalyzer #227: Specification Clarification

costastep, 2025-03-31:
Hi, I'm trying to understand/imagine how this device specifications compare to my now very old, over 15 years old, Logic Analyzer and more specifically in terms of how many samples it can capture, i.e. what size memory it uses to store the samples and how much memory is available/allocated per channel?

I was not able to find an answer by myself - the Wiki Documentation talks about "Maximum post-trigger samples", but that only raises more questions to me, e.g. is that number at 100MHz and for example at 50MHz we should multiply the Maximum by 2, at 25Mhz by 4, etc?

so, to make my thinking and hopefully not entirely stupid question more clear - my now very old device has 32 Mbits of SRAM and 32 channels, hence it uses (allocates) 1 Mbits per channel and that for example allows it to capture roughly 104.858 milliseconds of samples at 10Mhz (when no RLE compression is in use): 

1024 * 1024 / 10 000 000 = 0.104858ms

I read RP2350 has 520KB of SRAM, that is roughly 4 Mbits - is that is the memory used for storing the samples?

Thank you,
Costa

closed 2025-03-31 by gusmanb

costastep, 2025-03-31:
thank you for your very fast answer, but it seems to me I still need few clarifications:

> A sample is a sample, no matters which capture frequency you use, 

but if you have a fixed memory buffer size the frequency matters for how many samples you can store there, correct or not?

> in 8 channel mode (with a Pico2) you can capture 393.215 samples

is that "." a decimal point and you mean 393 samples only?! 

I am totally confused by such a small number and let me ask in another way on what I am wondering - for example:

1) if you want to Reverse-Engineer I2C communication that happens in 10 seconds time, because the Master and Slave talk from time to time with short messages, can you do it with this device, for example using the aforementioned "Burst Mode"? yes/no

2) and another scenario - can you capture continuously by that i mean 3-4 minutes I2S (let say 24bit/176.4Khz) and recover all the sound data? yes/no

gusmanb, 2025-03-31:
> thank you for your very fast answer, but it seems to me I still need few clarifications:
> 
> > A sample is a sample, no matters which capture frequency you use,
> 
> but if you have a fixed memory buffer size the frequency matters for how many samples you can store there, correct or not?

No, again, a sample is a sample, independent of the frequency, a sample for 8 channels is 8 bits, that's it. What changes is the amount of time, not the amount of samples.

> 
> > in 8 channel mode (with a Pico2) you can capture 393.215 samples
> 
> is that "." a decimal point and you mean 393 samples only?!

Sorry, is the thousands separator :)

> I am totally confused by such a small number and let me ask in another way on what I am wondering - for example:
>
>     1. if you want to Reverse-Engineer I2C communication that happens in 10 seconds [...]

Yes, set the trigger to be the SDA line going low (the first that happens in a start condition) and that's it, it will retrigger itself each time a new message is transmitted.

>     2. and another scenario - can you capture continuously by that i mean 3-4 minutes I2S (let say 24bit/176.4Khz) and recover all the sound data? yes/no

Obviously not.

costastep, 2025-03-31:
> a sample for 8 channels is 8 bits, that's it. What changes is the amount of time, not the amount of samples.

many thanks for the whole answer, now I believe it's entirely clear to me :)