gusmanb/logicanalyzer #230: Lockup on error dialog

derekfountain, 2025-04-02:
I'm not absolutely sure what I saw here, but I think it's worth investigating.

When setting up a capture it's possible to get an error dialog pop up. I forget exactly what I did wrong, I don't think it matters. What matters is that the buttons to Capture and Repeat Last Capture remained active while the error dialog was on screen. Because that error dialog popped up on a different screen I didn't see it, and clicked Capture. The user interface then locked up. I couldn't clear the error dialog, and I couldn't abort the capture. I think it deadlocked on the user buttons. I had to kill the whole program.

Sorry for the vague report. I think the fix is to disable the rest of the user interface while the error dialog is on screen?

gusmanb, 2025-04-02:
Without more info this is impossible to reproduce, I have tried to generate errors on captures, get a warning and capture again and there is no lock. Can you provide more info on which error did you received?

derekfountain, 2025-04-02:
Unfortunately I can't reproduce it now, either. I'll close the report and reopen it if I find a way to reproduce.

closed 2025-04-02 by derekfountain