gusmanb/logicanalyzer #237: "Index was outside the bounds of the array" when saving burst mode

rianadon, 2025-04-17:
Thanks for building this amazing logic analyzer! Very impressive you've written both the firmware and software for it.

I'm trying to import one of the `.lac` files I've saved, and I get this error upon importing.

[image]

Below are the settings I used to generate it. My suspicion is that this has something to do with burst mode and the "measure delays" feature.

[image]

gusmanb, 2025-04-17:
Hi. Could you upload the capture file that provokes the error?

Cheers.

rianadon, 2025-04-17:
Yup! GitHub didn't let me directly upload the `.lac`, so I added `.json` to the extension. The file is otherwise the same.

[capture.lac.json](https://github.com/user-attachments/files/19798530/capture.lac.json)

gusmanb, 2025-04-17:
Perfect, thanks, I will analyze it :)

gusmanb, 2025-04-17:
Ok, found the error, when saving a capture with burst mode enabled the number of bursts and hence the number of samples is saved incorrectly. It will be corrected in next version.