gusmanb/logicanalyzer #239: Version 6.0.0. Logic analyzer crash on Ubuntu 24.04.02

naith, 2025-04-18:
The latest version is 6.0.0. There are scratches after showing the logo, caused by SIGSEGV. 

[image]

[PythonInitLog.txt](https://github.com/user-attachments/files/19810876/PythonInitLog.txt)

Older version 5.1 fork fine

[image]

gusmanb, 2025-04-18:
It seems more related to your Python installation than from the analyzer. The analyzer does not use at all Python+GTK, it only uses Python to execute the analyzers and are just simple code without a GUI. Your log traces a warning about PyGI, that seems to be a package for Python to interact with Gnome.

Try to create a venv with Python 3.13 and launch the analyzer once it is activated, that may solve your problem.

naith, 2025-04-18:
Thank you for answering.

I added .venv environment and activated it, but I get an error message Python library not found.
The program runs, but without the decoder. It wants to add a path to python.cfg, but I don't know the path format and what. Please let me know what dependencies I need to meet to make it work.

Error message:
Cannot load Sigrok decoders. Make sure Python is installed on your computer. If, despite being installed, you still have problems, you can specify the path to the Python library in "python.cfg".

gusmanb, 2025-04-18:
There are no dependencies, a plain base installation is enough, you will need to point to the python library in the python.cfg, just that, the full path in a single line. The library is the .so one, as an example for python 3.7 it is "libpython3.7m.so", for 3.8 is "libpython3.8.so" and so on.

Also, you can check the “PythonInitLog.txt” file that will be created, it will contain a log of the detection procedure.

naith, 2025-04-18:
Another error:

18.04.2025 15:14 -> Initializing python system...
18.04.2025 15:14 -> Reading python path from config file...
18.04.2025 15:14 -> Stablished path: /usr/lib/x86_64-linux-gnu/libpython3.12.so

18.04.2025 15:14 -> Initializing decoders...
18.04.2025 15:14 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99

Startup script:
#! /bin/bash
echo "Logic Analyzer start ..."
source ./.venv/bin/activate
./LogicAnalyzer
echo "Done ..."

python.cfg:
/usr/lib/x86_64-linux-gnu/libpython3.12.so

gusmanb, 2025-04-18:
You are still trying to use your 3.12 installation, try with python 3.13 as directed in first response. Try first a clean 3.13 installation and if it still fails then with venv.

naith, 2025-04-18:
Thank you, but does not help. I still see the same error.
I tried to run clean Python3.13 (16:11) and venv (16:13), but I get the same result:

18.04.2025 16:11 -> Initializing python system...
18.04.2025 16:11 -> Reading python path from config file...
18.04.2025 16:11 -> Stablished path: /usr/lib/python3.13/config-3.13-x86_64-linux-gnu/libpython3.13.so

18.04.2025 16:11 -> Initializing decoders...
18.04.2025 16:11 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99
18.04.2025 16:13 -> Initializing python system...
18.04.2025 16:13 -> Reading python path from config file...
18.04.2025 16:13 -> Stablished path: /usr/lib/python3.13/config-3.13-x86_64-linux-gnu/libpython3.13.so

18.04.2025 16:13 -> Initializing decoders...
18.04.2025 16:13 -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99

gusmanb, 2025-04-18:
Try removing the cfg file, if you have update to python 3.13 it should pick it automatically from the environment variables (also, any whitespace, carriage return or similar in the cfg file may break it).

syabyr, 2025-05-22:
ubuntu 22.04 LTS:

without python.cfg:
5/22/2025 7:35 PM -> Initializing python system...
5/22/2025 7:35 PM -> Initializing python installation detection...
5/22/2025 7:35 PM -> Target Interpreter: python,  Initializing Python Installation Detection...
5/22/2025 7:35 PM -> Target Interpreter: python,  Python version found: 313
5/22/2025 7:35 PM -> Target Interpreter: python,  Python version parsed: 3.13
5/22/2025 7:35 PM -> Target Interpreter: python,  PythonExecutable, Finding python library on Mac/Linux...
5/22/2025 7:35 PM -> Target Interpreter: python,  PythonExecutable, Script output: 
5/22/2025 7:35 PM -> Target Interpreter: python,  Final python path: 
5/22/2025 7:35 PM -> Target Interpreter: python,  Cannot retrieve Python library path, aborting.
5/22/2025 7:35 PM -> Target Interpreter: python3,  Initializing Python Installation Detection...
5/22/2025 7:35 PM -> Target Interpreter: python3,  Python version found: 313
5/22/2025 7:35 PM -> Target Interpreter: python3,  Python version parsed: 3.13
5/22/2025 7:35 PM -> Target Interpreter: python3,  PythonExecutable, Finding python library on Mac/Linux...
5/22/2025 7:35 PM -> Target Interpreter: python3,  PythonExecutable, Script output: 
5/22/2025 7:35 PM -> Target Interpreter: python3,  Final python path: 
5/22/2025 7:35 PM -> Target Interpreter: python3,  Cannot retrieve Python library path, aborting.
5/22/2025 7:35 PM -> Valid installations found: 0
5/22/2025 7:35 PM -> No valid python installation found, aborting startup.

with python.cfg:
5/22/2025 7:26 PM -> Initializing python system...
5/22/2025 7:26 PM -> Reading python path from config file...
5/22/2025 7:26 PM -> Stablished path: /home/huiwang/anaconda3/envs/py313/lib/libpython3.13.so

5/22/2025 7:26 PM -> Initializing decoders...
5/22/2025 7:26 PM -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99