gusmanb/logicanalyzer #247: Include more Sigrok Decoders (i.e. mos6502)

Synthron, 2025-05-07:
Hi. 

I want to use the LogicAnalyzer for my homebrew 6502 computer project and there are available Sigrok Decoders [here](https://github.com/hoglet67/libsigrokdecode/tree/master/decoders/mos6502)

Including them doesn't seem as easy as copy-pasting the files into the decoder-folder, so how can I include them?
I looked into your documentation on how to implement own decoders but I don't seem to get what to do exactly. 

grettings

Synthron

gusmanb, 2025-05-08:
Well, it is, I just tested the one you linked and it works... Copy the folder "mos6502" from that repo to the "decoders" folder (do not place the files on the decoders folder itself), restart the program and check under "All" as the decoder does not have a proper tag for its own category.

[image]

Synthron, 2025-05-08:
thanks, got it working somehow (didn't know what I did wrong the first time)
but seems like the decoder is not working as I want it to, I guess I have to modify and fix it myself...

for now, issue can be closed, thanks :)

closed 2025-05-08 by Synthron