gusmanb/logicanalyzer #250: decoders starting with numbers not recognized

Synthron, 2025-05-16:
I wrote a decoder with the name "65c02" and put it inside a folder with the same name inside the decoders directory.
It didn't show up inside the LogicAnalyzer software, no matter what. 

Then I renamed it to "wdc65c02" and voila, it was available.

Somehow decoders starting with a number are not included in the software. Is this a bug or an intended feature?

best regards

gusmanb, 2025-05-17:
Python package names cannot start by a number: https://stackoverflow.com/questions/17487209/can-a-python-package-name-start-with-a-number

closed 2025-05-17 by gusmanb

Synthron, 2025-05-17:
thanks for clarifying. I was just wandering because for troubleshooting purposes I tested the package with sigrok-cli and it showed up perfectly fine, being usable as well. 

looks like sigrok uses a different include method for packages starting with numbers then....