gusmanb/logicanalyzer #251: Scroll bar ratio seemed has some problem

2jiangjiang, 2025-05-26:
When I use the LogicAnalyzer 6.0 software open a  lac file which saved by itself. The scroll bar at the bottom have a ratio problem,it case can't visit the latter part of data

[image]

the screenshot shows the actual data is lager than the scroll bar range

gusmanb, 2025-05-26:
Hmmm, cannot reproduce it. 
Could you share the .lac file? 
Also, which OS are you using?

2jiangjiang, 2025-05-27:
[usb.lac.json](https://github.com/user-attachments/files/20447041/usb.lac.json)
Okay this is my lac file.
And my OS is Windows 11 professonal 24H2 version 26100.4061.

2jiangjiang, 2025-05-27:
I found where cause the issue 
I use Burst mode to catch samples,the scroll bar range seems decision by "PostTriggerSamples",but lac file record only once "PostTriggerSamples".
Scroll bar range in once Burst mode catched

gusmanb, 2025-05-27:
Ok, that makes sense, there is a bug in the saving of a file that uses burst mode, it is solved in the next release. Will try to send you a beta of the new version this weekend.