gusmanb/logicanalyzer #262: Support for the Raspberry 2 Pico W

brainiac51, 2025-06-19:
While going through the different supportet versions of Raspberry Picos, I noticed that the Pico 2 W is not stated as supported, but the normal Pico W is. Is the Pico 2 W also compatible and if not, will it be supported in the future?

gusmanb, 2025-06-19:
Hi. Is not compatible right now, the V6.5 that I'm preparing adds support for it, I hope to release it soon :)

MaikuMori, 2025-07-25:
Maybe we can keep the issue active to track this?

I see there's a 6.5 branch. How finished is that in terms of Pico 2 W support?

gusmanb, 2025-07-25:
> Maybe we can keep the issue active to track this?
> 
> I see there's a 6.5 branch. How finished is that in terms of Pico 2 W support?

https://github.com/gusmanb/logicanalyzer/discussions/277