gusmanb/logicanalyzer #263: LogicAnalyzer crashes during "Initializing decoders"

TISteveDC, 2025-06-20:
When running LogicAnalyzer I get the splash screen then after a few seconds it disappears and nothing else happens.
In Task manager it appears, then also disappears.
The log file indicates it reached "Initializing decoders" but there are no further entries in the log file after this.

6/20/2025 1:51 PM -> Initializing python system...
6/20/2025 1:51 PM -> Initializing python installation detection...
6/20/2025 1:51 PM -> Target Interpreter: python,  Initializing Python Installation Detection...
6/20/2025 1:51 PM -> Target Interpreter: python,  Python version found: 37
6/20/2025 1:51 PM -> Target Interpreter: python,  Python version parsed: 3.7
6/20/2025 1:51 PM -> Target Interpreter: python,  Finding python library on Windows...
6/20/2025 1:51 PM -> Target Interpreter: python,  Script output: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Target Interpreter: python,  Final python path: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Target Interpreter: python,  Python path found: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Target Interpreter: python3,  Initializing Python Installation Detection...
6/20/2025 1:51 PM -> Target Interpreter: python3,  Python version not found, aborting.
6/20/2025 1:51 PM -> Valid installations found: 1
6/20/2025 1:51 PM -> Selected version: 3.7
6/20/2025 1:51 PM -> Stablished path: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Initializing decoders...
6/20/2025 1:51 PM -> Initializing python system...
6/20/2025 1:51 PM -> Initializing python installation detection...
6/20/2025 1:51 PM -> Target Interpreter: python,  Initializing Python Installation Detection...
6/20/2025 1:51 PM -> Target Interpreter: python,  Python version found: 37
6/20/2025 1:51 PM -> Target Interpreter: python,  Python version parsed: 3.7
6/20/2025 1:51 PM -> Target Interpreter: python,  Finding python library on Windows...
6/20/2025 1:51 PM -> Target Interpreter: python,  Script output: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Target Interpreter: python,  Final python path: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Target Interpreter: python,  Python path found: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Target Interpreter: python3,  Initializing Python Installation Detection...
6/20/2025 1:51 PM -> Target Interpreter: python3,  Python version not found, aborting.
6/20/2025 1:51 PM -> Valid installations found: 1
6/20/2025 1:51 PM -> Selected version: 3.7
6/20/2025 1:51 PM -> Stablished path: C:\Anaconda3\Python37.dll
6/20/2025 1:51 PM -> Initializing decoders...

TISteveDC, 2025-06-20:
Tried with and without the new line, and with both "/" and "\" separators. Also tried with and without the separator at the end of the path. Same result.

I always see a blank line after the "Stablished path" in the log. If there is a blank line in Python.cfg then there are TWO blank lines in the log.

gusmanb, 2025-06-20:
You still have a newline if you see that blank, maybe your text editor is messing it up.

This is a log that I have created right now placing in my cfg your path:

```
20/06/2025 22:26 -> Initializing python system...
20/06/2025 22:26 -> Reading python path from config file...
20/06/2025 22:26 -> Stablished path: C:\TEMP\LogicAnalyzer\Python32
20/06/2025 22:26 -> Initializing decoders...
```

As you can see, no newline is shown in the log.

[Python.cfg.txt](https://github.com/user-attachments/files/20842795/Python.cfg.txt)

I'm attaching the cfg file, remove the .txt estension and try it, it contains your path.

TISteveDC, 2025-06-20:
Strange. I am using VIM as my editor so not sure what is going on with the end of line.

Either way, with your file I still get the error window. Logfile doesn't have the blank line(s) now though, so progress I guess. LOL 

I will try on a different computer at home to see what happens there.

Just want to say a huge thanks for the quick responses. I truly appreciate every second of your time :)

6/20/2025 3:33 PM -> Initializing python system...
6/20/2025 3:33 PM -> Reading python path from config file...
6/20/2025 3:33 PM -> Stablished path: C:\TEMP\LogicAnalyzer\Python32
6/20/2025 3:33 PM -> Initializing decoders...
6/20/2025 3:33 PM -> Error initializing python engine: The type initializer for 'Delegates' threw an exception. -    at Python.Runtime.Runtime.Delegates.get_Py_IsInitialized()
   at Python.Runtime.Runtime.Py_IsInitialized()
   at Python.Runtime.Runtime.<>c.<Initialize>b__32_0()
   at Python.Runtime.Runtime.TryUsingDll[T](Func`1 op)
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize()
   at SigrokDecoderBridge.SigrokPythonEngine.EnsureInitialized() in F:\LogicAnalyzerRepo\Software\LogicAnalyzer\LogicAnalyzer\SigrokDecoderBridge\SigrokPythonEngine.cs:line 99

gusmanb, 2025-06-20:
Read my last response, you're missing the DLL name :)