gusmanb/logicanalyzer #270: Program crashes when entering network settings

Emile666, 2025-07-07:
First of all, thanks for an excellent program, it is very impressive! I want to use 2 boards for a 6502 Logic Analyzer, something similar as I have seen for the ZX spectrum. I need to write a 6502 Sigrol protocol analyzer, but first thing is to get your program to work.

I have two RP2040 Pico W-2023 and used firmware logic-analyzer-firmware_v6.0.0_BOARD_PICO_W_WIFI.zip.
The Logic Analyzer program starts correctly, but when I enter Access-point (WOffice, which is an Apple air-play access point), password (upper/lower case and special chars like a '!') and address (192.168.1.75, I verified in the router at 192.168.1.1 that this one was free) and press Accept, the program crashes and returns to Windows with no messages given.

I tried to enter the IP address directly instead of the WOffice name, but that didn't work. I also tried to input a password without any special character, but again, the program keeps on crashing. I tried using the PICO_W firmware (without the _WIFI), but then network settings are no longer available (which makes sense).

Can you help me out here? I might be doing something wrong, but don't know what it is.

Image is from the Pico I am currently using.
[image]

gusmanb, 2025-07-07:
Hi.

I'm sorry to tell you but that's not a Pico-W, that's a chinese knock-off that is sold like a Pico-W but it does not include the Cypress WiFi module but an ESP2866 modem that is completelly incompatible...

Sorry.

Emile666, 2025-07-07:
Thanks for the quick reply! Well, that would explain it, wouldn't it? I didn't know that there are multiple Pico W versions out there. I am going to buy a few original Pico-W PCBs and will try again. On the other hand, it would be nice if the program gave an error message (incompatible Pico HW version or something) instead of just crashing.

gusmanb, 2025-07-07:
I'm not sure why it's crashing as I don't have one of these to test it. If you are using Windows you can check the event viewer and there you will see the crash information, with that I may be able to determine the problem.

gusmanb, 2025-07-07:
Perfect. I think I know what it is, the device is not responding and the timeout is causing the crash, I will protect the network config functions against timeout. Thanks :)

Emile666, 2025-07-08:
Just to let you know: I installed two original Pico W and now everything is working fine. I have 48 channels available and Wifi is working fine, impressive!